dh1tw/remoteAudio #2: Raspberry Pi B+ "Illegal instruction"

dseven, 2017-02-05:
Should the v0.1.2 ARM build run on a Raspberry Pi B+ ? I get:
```
pi@raspberrypi:~ $ ./remoteAudio
Illegal instruction
pi@raspberrypi:~ $ file remoteAudio
remoteAudio: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 2.6.32, BuildID[sha1]=d762b74ccb3d156bf0c365967348d08ea982ad87, not stripped
pi@raspberrypi:~ $ cat /proc/cpuinfo
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 697.95
Features        : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2708
Revision        : 0010
Serial          : 00000000738b6f40
pi@raspberrypi:~ $ ldd remoteAudio
        /usr/lib/arm-linux-gnueabihf/libarmmem.so (0xb6fb5000)
        libportaudio.so.2 => /usr/lib/arm-linux-gnueabihf/libportaudio.so.2 (0xb6f7e000)
        libasound.so.2 => /usr/lib/arm-linux-gnueabihf/libasound.so.2 (0xb6ea4000)
        libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0xb6e29000)
        libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0xb6e01000)
        libsamplerate.so.0 => /usr/lib/arm-linux-gnueabihf/libsamplerate.so.0 (0xb6c8e000)
        libopus.so.0 => /usr/lib/arm-linux-gnueabihf/libopus.so.0 (0xb6c40000)
        libopusfile.so.0 => /usr/lib/libopusfile.so.0 (0xb6c26000)
        libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0xb6ae5000)
        libjack.so.0 => /usr/lib/arm-linux-gnueabihf/libjack.so.0 (0xb6ac0000)
        librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0xb6aa8000)
        /lib/ld-linux-armhf.so.3 (0x7f5b4000)
        libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0xb6a95000)
        libogg.so.0 => /usr/lib/arm-linux-gnueabihf/libogg.so.0 (0xb6a87000)
        libdb-5.3.so => /usr/lib/arm-linux-gnueabihf/libdb-5.3.so (0xb68ff000)
pi@raspberrypi:~ $ 
```

tehmaze, 2017-02-06:
You are trying to execute a 32-bit Intel binary on tour Raspberry Pi, that won't work

dh1tw, 2017-02-06:
@tehmaze is correct. Please download and use the ARM binary from the [release page|https://github.com/dh1tw/remoteAudio/releases]. If you then still have problems, feel free to open  another issue!

closed 2017-02-06 by dh1tw

dseven, 2017-02-06:
I guess neither of you read the output of the "file" command ... or I'm losing it....

dh1tw, 2017-02-06:
Right - sorry. In principal it should work. I compiled the binary on a Raspberry Pi 3. Did you compile the binary (if - which version of Go did you use) or did you download the compiled binary?

dseven, 2017-02-06:
I downloaded https://github.com/dh1tw/remoteAudio/releases/download/v0.1.2/remoteAudio-v0.1.2-linux-arm.tgz . I suspect that the Raspberry Pi 3 uses a newer version of the ARM instruction set, and the binaries may be incompatible. I'll try to find time to get a development environment together. I did get the x86_64 build to work between a couple of boxes running Fedora at home, but not sure I want to dedicate either of them to the "hilltop end" for a real remote trial - the Pi is not doing anything, so it'd be nice if that could be made to work..

dh1tw, 2017-02-06:
I know that Go 1.9 will drop support for older ARM Platforms, but it could indeed be possible that there have been changes in the ARM instruction sets between Raspi (Model B) and Raspi 3. I unfortunately don't have an Raspberry Pi 1 anymore in order to give it a try. But it should be possible to get remoteAudio compiled on the Raspi 1 - at least with Go 1.7 / 1.8rc.

Please let me know if you need assistance during the build process!

dseven, 2017-02-07:
I got go 1.7.5 installed on the Pi, but hit a wall trying to build protobuf - it's failing when trying to link libprotobuf.so with no useful error message - only "ld returned 1 exit status". It's starting to feel like more trouble than it's worth. Maybe I'll find more enthusiasm another day....

dseven, 2017-02-07:
Well, I got it to build (protobuf took hours). Now to get it to open an input device ....

```
root@raspberrypi:~# remoteAudio enumerate
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.front.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround40.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround40
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround41
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround50
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround51
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround71.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround71
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM iec958
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline

Available audio devices and supported Host APIs:

        Detected 2 host API(s):

        Name:                   ALSA
        Default input device:   iMic USB audio system: Audio (hw:1,0)
        Default output device:  default
        Devices:
                Name:                      bcm2835 ALSA: - (hw:0,0)
                MaxInputChannels:          0
                MaxOutputChannels:         2
                DefaultLowInputLatency:    -1s
[... 87 lines cut ...]
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
[... cut ...]
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
2017/02/07 21:22:19.332379 Connected to MQTT Broker nuc:1883
OutputUnderflow
unable to open recording audio stream on device iMic USB audio system: Audio (hw:1,0)
invalid StreamCallback
```

dseven, 2017-02-07:
Got it sortof working (by not trying to specify the input device) ... but showing high latency on the client (which is the x86_64 box that was working well in previous test, on the same LAN) - 200-400ms, getting jitter, and lots of "InputOverflow" reports on the (Pi) server. Maybe the Pi 1 is not up to the task :/

dh1tw, 2017-02-07:
Hey, 
all the alsa warnings you see are not critical. If you want to remove them you have to remove or uncomment them in `/usr/share/alsa/alsa.conf`. If no input/output sound device is specified to remoteAudio, then the default card will be used. In your case it seems to point to the built-in bcm2835.

You can explicitly set the default soundcard by adding $HOME/.asoundrc

```
pcm.!default {
        type hw
        card 1
        device 0
}

ctl.!default {
        type hw
        card 1
        device 0
}
```

This requires a reload of Alsa or a simple restart of the Raspberry Pi.

Have you checked the CPU usage on the PI with something like `htop` or `top` ? 

With which version of Go did you compile remoteAudio? 1.7 or 1.8rc ? 

OutputOverflows mean that portaudio/alsa was not able to write the data in time to the sound card. This might be an issue of not enough CPU power available. That could also be the reason for the extremely high latency. I have on a local network a latency of 1ms with a Raspi 3.

Have you tried to increase the rx buffer size?