dh1tw/remoteAudio #5: Add pflags to client for starting the audio streaming (rx and tx)

dh1tw, 2017-03-06:
Currently both audio streams (rx and tx) have to be started and stopped through the web interface. It would be handy if both could be started through a CLI plfag when executing the client.

On Exit the TX Audio will be stopped. And the RX Audio will timeout automatically when #4 is implemented. 

Stopping the RX Audio on exit is not an option as more clients might still be listening.

przemorx, 2017-11-09:
Add one more vote for this enhancement - it is very annoying to always turn audio on.
Mek SQ3RX

dh1tw, 2019-11-15:
this has been added now to v0.4.0. When starting the remoteAudio client you can tell an existing remoteAudio server to start streaming. Example:
`$ remoteAudio client nats --stream-on-startup --server-name ts480`

Another way to start / stop streams without using the web interface is by making use of the [REST interface](https://github.com/dh1tw/remoteAudio/wiki/Integrate-with-3rd-party-Software-(REST-Interface)). This is a great way to integrate remoteAudio with other applications.

closed 2019-11-15 by dh1tw