dh1tw/remoteAudio #13: Feature request: 2 new topics

MichaelRenner, 2017-09-02:
Moin Tobias,

I'm looking for a way to control the PTT of the tranceiver. A simple possibility would be: remoteAudio creates two new topics (for example, control/rx and control/tx) containing the state of the buttons.
On the server side, an additional script can subscribe to the topic and call rigctrl accordingly.

This would be very helpful, the NF vox is often too slow.

73 de dd0ul

dh1tw, 2017-10-31:
Hi @MichaelRenner,

I was thinking about adding a REST endpoint on the client side. So any local application could query through a JSON Message the status of the audio streams and control them.

Adding an MQTT is certainly also possible, but when the broker is on the WAN this would add the latency of another round trip. Therefore I think the local rest endpoint might be better. 

how?

73 Tobias

MichaelRenner, 2017-11-01:
On 31.10.2017 10:33, Tobias Wellnitz wrote:
>
> Hi @MichaelRenner <https://github.com/michaelrenner>,
>

Moin Tobias,

thanks for comming back to my proposal.

> I was thinking about adding a REST endpoint on the client side. So any
> local application could query through a JSON Message the status of the
> audio streams and control them.
>

REST is a common protocol and easy to use - regardless of the operating
system.

> Adding an MQTT is certainly also possible, but when the broker is on
> the WAN this would add the latency of another round trip. Therefore I
> think the local rest endpoint might be better.
>

That will work for sure. Which functions did you think of? Without
extension on the server side, the REST interface can only provide the
functions that the client offers on port 8080, right?

BTW: Do you know a mqtt server in the HamNet?

73, Micha

-- 
|Michael Renner      E-mail: michael.renner@gmx.de  |
|81541 Munich        Twitter: @dd0ul                |
|Germany             Don't drink as root!      ESC:wq

dh1tw, 2019-11-14:
Hi @MichaelRenner,
some time has passed by since you opened this issue ;-) I've just published version 0.4.0 of remoteAudio. You can fetch a prebuild binary from the [releases page](https://github.com/dh1tw/remoteAudio/releases). 

since version 0.3.0 remoteAudio includes now a REST interface which let's you conveniently control remoteAudio from different applications. I've written a dedicated [wiki article on how to integrate remoteAudio with 3rd party applications](https://github.com/dh1tw/remoteAudio/wiki/Integrate-with-3rd-party-Software-(REST-Interface)).

73 Tobias
DH1TW

closed 2019-11-14 by dh1tw