dh1tw/remoteAudio #14: server disconnect, manual restart is needed

federicobriata, 2018-02-17:
Hi Tobias
I made a systemd service to start a server but after a while it's stop to steam and I've to restart the service. This is what I get from the status.

Feb 17 10:22:19 beaglebone remoteAudio_server.sh[1470]: 2018/02/17 10:22:19.606643 Connection lost to MQTT Broker; Reason: pingresp not received, disconnecting
Feb 17 10:22:19 beaglebone remoteAudio_server.sh[1470]: 2018/02/17 10:22:19.683039 Connected to MQTT Broker myserver.com:1234
Feb 17 10:22:31 beaglebone remoteAudio_server.sh[1470]: OutputUnderflow
Feb 17 10:22:37 beaglebone remoteAudio_server.sh[1470]: 2018/02/17 10:22:37.684099 Connection lost to MQTT Broker; Reason: pingresp not received, disconnecting
Feb 17 10:22:37 beaglebone remoteAudio_server.sh[1470]: 2018/02/17 10:22:37.759613 Connected to MQTT Broker myserver.com:1234
Feb 17 10:22:50 beaglebone remoteAudio_server.sh[1470]: InputOverflow
Feb 17 10:22:55 beaglebone remoteAudio_server.sh[1470]: 2018/02/17 10:22:55.761074 Connection lost to MQTT Broker; Reason: pingresp not received, disconnecting
Feb 17 10:22:55 beaglebone remoteAudio_server.sh[1470]: 2018/02/17 10:22:55.839860 Connected to MQTT Broker myserver.com:1234
Feb 17 10:23:13 beaglebone remoteAudio_server.sh[1470]: 2018/02/17 10:23:13.841402 Connection lost to MQTT Broker; Reason: pingresp not received, disconnecting
Feb 17 10:23:13 beaglebone remoteAudio_server.sh[1470]: 2018/02/17 10:23:13.943125 Connected to MQTT Broker myserver.com:1234

beside this, remoteaudio it's very cool! :)
73 iz1glg, federico

federicobriata, 2018-02-17:
$ remoteAudio version
remoteAudio Version: 0.2.1-11-gacdd864, linux/arm, BuildDate: 2018-02-17T10:39:13Z, Commit: acdd864

federicobriata, 2018-02-17:
$ cat /lib/systemd/system/remoteaudio.service
[Unit]
Description=remoteaudio
After=sshd.service alsa-restore.service

[Service]
Type=simple
TimeoutSec=infinity
User=debian
Group=debian
ExecStart=/home/debian/remoteAudio_server.sh
#ExecStartPost=/usr/bin/chrt -p 80 $MAINPID
ExecStartPost=/usr/bin/renice 10 -p $MAINPID

[Install]
WantedBy=multi-user.target

$ cat /home/debian/remoteAudio_server.sh
#!/bin/bash
/usr/bin/remoteAudio server mqtt -u myserver.com -p 1234 -U mytest -P test12345 --input-device-latency 10ms --output-device-latency 10ms

dh1tw, 2018-02-19:
Hey Frederico,
well, it seems that the connection to the broker is lost. I presume that the server name `myserver.com:1234` has been redacted. Is your connection otherwise stable which you have to that server? 

You could add to your systemd service an automatic restart policy. 

```
[Service]
Restart=always
RestartSec=5 #try again after 5 sec
```

federicobriata, 2018-02-20:
Ciao Tobias,
maybe for a while the connection to myserver.com:1234 was lost, because I reach internet from remote wifi BTS.

unfortunately systemd restart policy take care only when the process exit, but in my case the status the process was still up and running, remoteaudio just stop to retry to reconnect.
There is a limit on connection retry and if so can be changed to retry always?

dh1tw, 2018-02-20:
Ok, so if I understand correctly, despite that the connection to the broker got lost, the application doesn't quit, right?
I think quitting after failure is the better strategy. Systemd is the better tool to handle retries.

I'm anyway rewriting most of the code these days but I will keep this this particular case in mind and make sure that the application exits properly.

federicobriata, 2018-03-04:
ok sounds good. thanks.

dh1tw, 2019-11-14:
With remoteAudio 0.4.0 we have moved from MQTT to NATS as our standard transport. The lastest version contains a lot of bug fixes and has proven to be stable. You can fetch some prebuilt binaries from the [release page](https://github.com/dh1tw/remoteAudio/releases).

closed 2019-11-14 by dh1tw