dh1tw/remoteAudio #16: Cannot obtain info for CTL elem

kovalroma, 2018-12-17:
Linux Mint 
./remoteAudio server nats

```
Using config file: /home/roman/.remoteAudio.toml
ALSA lib setup.c:548:(add_elem) Cannot obtain info for CTL elem (MIXER,'AC97 2ch->4ch Copy Switch',0,0,0): No such file or directory
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib setup.c:548:(add_elem) Cannot obtain info for CTL elem (MIXER,'AC97 2ch->4ch Copy Switch',0,0,0): No such file or directory
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib setup.c:548:(add_elem) Cannot obtain info for CTL elem (PCM,'IEC958 Playback PCM Stream',0,0,0): No such file or directory
ALSA lib setup.c:548:(add_elem) Cannot obtain info for CTL elem (PCM,'IEC958 Playback PCM Stream',0,0,0): No such file or directory
ALSA lib setup.c:548:(add_elem) Cannot obtain info for CTL elem (PCM,'IEC958 Playback PCM Stream',0,0,0): No such file or directory
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for 4294967295, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for 4294967295, skipping unlock
2018/12/17 16:11:28 broker:nats: no servers available for connection

```
Can someone help me and tell what's wrong ? Something with Alsa sound card, but what exactly ?

dh1tw, 2018-12-18:
Hi @kovalroma,

I suspect there are two separated issues in your log. There are some issues with your sound device and you apparently try to connect to an non exisiting nats broker.

Sound card / alsa issues: 
1. Does the sound card work properly on your system? Try `speaker-test -t wav`. That should generate sound on your speakers. With `arecord test.wav` you can create a test recording, and replay it with `aplay test.wav`.
If that works, execute `./remoteAudio enumerate` to check if your soundcard is properly recognized. You can also check the wiki on remoteAudio for some further tips & tricks.

2. specify the nats broker. You can download gnatsd and run an instance on your local machine.

For debugging purposes it might be easier to call `remoteAudio` with pflags instead of using the config file, or add your config file to this issue.

kovalroma, 2018-12-18:
1. Something wrong with my sound card. When I play ` aplay   ./piano2.wav ` it shows that music is playing, but no sound in the speaker. Anyway I will fixed this issue by myself. 

2. nats broker is installed and running. Here is telnet to the server 
`telnet 192.168.19.92 4222
Trying 192.168.19.92...
Connected to 192.168.19.92.
Escape character is '^]'.
INFO {"server_id":"53Ga4xqZLv1Qh9Co6ROnwZ","version":"1.3.0","proto":1,"go":"go1.11","host":"0.0.0.0","port":4222,"max_payload":1048576,"client_id":28} 
`

My **remoteAudio.toml** config:
```

[nats]
broker-url = "192.168.19.92"
broker-port = 4222
username = ""
password = ""
radio = "ft950"

[input-device]
device-name = "default"
samplerate = 48000
latency = "5ms"
channels = 1

[output-device]
device-name = "default"
samplerate = 48000
latency = "5ms"
channels = 2

[opus]
application = "restricted_lowdelay"
bitrate = 24000
complexity = 5
max-bandwidth = "wideband"

[audio]
rx-buffer-length = 10 # 10x10ms (@framesize=480 & samplerate=48kHz)
# opus does not accept arbitrary frame length! Check OPUS documentation
frame-length = 480  #10ms @ 48kHz (should work for most cases)
tx-volume = 70
rx-volume = 70
stream-on-startup = false

[http]
host = "localhost"
port = 9090

```

I think that firstly I need to fixed my sound issues and later problem with `gnatsd`

kovalroma, 2018-12-19:
@dh1tw  I've tested on completely fresh Linux installation and it works. Thank you.

dh1tw, 2018-12-19:
Good to hear that remoteAudio works now as expected! I case you might want to dig into the code, please be adviced that the recent changes are still not merged into the master branch. They are located in the branch [natstest](https://github.com/dh1tw/remoteAudio/tree/natstest).

closed 2018-12-19 by dh1tw