dh1tw/remoteAudio #17: Build does not work

8cH9azbsFifZ, 2019-05-14:
$ make build

protoc --proto_path=./icd --gofast_out=./sb_audio ./icd/audio.proto
protoc --proto_path=./icd --micro_out=./sb_audio ./icd/audio.proto
protoc-gen-micro: program not found or is not executable
--micro_out: protoc-gen-micro: Plugin failed with status code 1.
make: *** [build] Error 1

dh1tw, 2019-05-14:
Indeed, the protoc-gen-micro package was missing. I updated the makefile and the [build instructions in the wiki](https://github.com/dh1tw/remoteAudio/wiki/Build-instructions-for-Linux). Now you should be able to build remoteAudio.

8cH9azbsFifZ, 2019-05-15:
One step further, but still build fails with:

`source $HOME/protobuf/bin/protobufrc`
` No such file or directory`

And
`make build
protoc --proto_path=./icd --gofast_out=./sb_audio ./icd/audio.proto
protoc-gen-gofast: program not found or is not executable
--gofast_out: protoc-gen-gofast: Plugin failed with status code 1.
`

dh1tw, 2019-05-15:
have you updated the repo? 
`go get -u` or `git pull`

dh1tw, 2019-11-14:
I have updated the build scripts and fixed the dependencies with go modules. You should now be able to compile remoteAudio with go 1.13 without any problems. If you don't want to compile remoteAudio you can download one of the freshly baked binaries of version 0.4.0 from [the releases page](https://github.com/dh1tw/remoteAudio/releases)

closed 2019-11-14 by dh1tw