dh1tw/remoteAudio #18: install-deps fails (xubuntu 18.10)

g0wfv, 2019-05-27:
As per title - install-deps fails on xubuntu 18.10

```
$ make install-deps
go get github.com/gogo/protobuf/protoc-gen-gofast
go get github.com/GeertJohan/go.rice/rice
go get github.com/micro/protoc-gen-micro
go get ./...
# github.com/dh1tw/remoteAudio/cmd
cmd/client_nats.go:169:47: cannot use brNatsOpts (type "github.com/nats-io/go-nats".Options) as type "github.com/nats-io/nats.go".Options in argument to "github.com/micro/go-plugins/broker/nats".Options
cmd/server_nats.go:154:47: cannot use brNatsOpts (type "github.com/nats-io/go-nats".Options) as type "github.com/nats-io/nats.go".Options in argument to "github.com/micro/go-plugins/broker/nats".Options
Makefile:47: recipe for target 'install-deps' failed
make: *** [install-deps] Error 2
```

dh1tw, 2019-05-27:
I will check and come back to you

dh1tw, 2019-05-27:
The root cause of this problem is that one of the dependencies (go-micro) has modified the nats plugin which remoteAudio is using for communication. I've submitted [two pull requests](https://github.com/micro/go-plugins/pulls) which will fix the problem. I will let you know when the PR get's accepted. Then you should be able to build remoteAudio.

g0wfv, 2019-05-28:
I had spotted 2 "similar looking" repos in the directory structure and went so far as to doing a diff on the directories ....

.... Then I got sidetracked and forgot to press it further!

Seems I was along the right lines just not far enough along those lines!

Thanks.  Will test and report back later.

g0wfv, 2019-05-28:
Those PRs seem to have made things worse ....
```
$ make install-deps 
go get github.com/gogo/protobuf/protoc-gen-gofast
go get github.com/GeertJohan/go.rice/rice
go get github.com/micro/protoc-gen-micro
go get ./...
# github.com/dh1tw/remoteAudio/cmd
cmd/client_nats.go:167:44: cannot use regNatsOpts (type "github.com/nats-io/go-nats".Options) as type "github.com/nats-io/nats.go".Options in argument to "github.com/micro/go-plugins/registry/nats".Options
cmd/client_nats.go:168:42: cannot use trNatsOpts (type "github.com/nats-io/go-nats".Options) as type "github.com/nats-io/nats.go".Options in argument to "github.com/micro/go-plugins/transport/nats".Options
cmd/client_nats.go:169:47: cannot use brNatsOpts (type "github.com/nats-io/go-nats".Options) as type "github.com/nats-io/nats.go".Options in argument to "github.com/micro/go-plugins/broker/nats".Options
cmd/server_nats.go:153:44: cannot use regNatsOpts (type "github.com/nats-io/go-nats".Options) as type "github.com/nats-io/nats.go".Options in argument to "github.com/micro/go-plugins/registry/nats".Options
cmd/server_nats.go:154:47: cannot use brNatsOpts (type "github.com/nats-io/go-nats".Options) as type "github.com/nats-io/nats.go".Options in argument to "github.com/micro/go-plugins/broker/nats".Options
cmd/server_nats.go:155:42: cannot use trNatsOpts (type "github.com/nats-io/go-nats".Options) as type "github.com/nats-io/nats.go".Options in argument to "github.com/micro/go-plugins/transport/nats".Options
Makefile:47: recipe for target 'install-deps' failed
make: *** [install-deps] Error 2
```
Also changing the URL in `cmd/client_nats.go` and `cmd/server_nats.go` seems to let it complete with no errors ...
```
$ make install-deps
go get github.com/gogo/protobuf/protoc-gen-gofast
go get github.com/GeertJohan/go.rice/rice
go get github.com/micro/protoc-gen-micro
go get ./...
```
I'll test the rest of the build and submit a PR if successful ....

g0wfv, 2019-05-28:
PR submitted - build now succeeds.

closed 2019-05-28 by g0wfv

dh1tw, 2019-05-28:
perfect. Since @asim already merged the two PRs, you have to update all dependencies with
`go get -u ./...`

g0wfv, 2019-05-28:
@dh1tw You need to accept my PR #19 to this repo to finish the fix ....

dh1tw, 2019-05-28:
already merged! If you have any further issues, please let me know! I'm keen on receiving feedback!

g0wfv, 2019-05-28:
Just trying fist tests and to see what it can do ...... I *think* i've got this right .....