dh1tw/remoteAudio #21: make install-dups fails on crypto/ed25519

K6TD, 2019-10-13:
`make install-deps
go get github.com/gogo/protobuf/protoc-gen-gofast
go get github.com/GeertJohan/go.rice/rice
go get github.com/micro/protoc-gen-micro
go get ./...
package crypto/ed25519: unrecognized import path "crypto/ed25519" (import path does not begin with hostname)
make: *** [Makefile:50: install-deps] Error 1`
============================================================
`cat os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs`
======================================================

dh1tw, 2019-10-13:
Hey Kevin, 
I'm on the road right now. I'll have a more detailed look into the error when I'm back home. I think that upgrading to go 1.13 should fix the issue. 
73 Tobias

K6TD, 2019-10-13:
I suspect you are right.  Lots of notes in gopher about the change around that package.

TKS.

73, K6TD

K6TD, 2019-10-13:
go1.13.1 cures the issue with crypto/ed25519, but, a new issue occurs with cores/etcd/clientv3.

see -https://github.com/etcd-io/etcd/pull/10044#commitcomment-30334256, which i think is an issue with how the modules are referenced in remoteAudio.

dh1tw, 2019-10-14:
Hi Kevin,
recently micro (the microservice framework I'm for remoteAudio) has changed from consul to etcd as the default registry. Apparently the [issue has already been raised](https://github.com/micro/go-micro/issues/851). I guess I have to start using go modules to fix the versions of the dependencies. These kind of annoying errors are happening now more often. Typically the micro team is pretty responsive and fixes these issues within a day or two, but introducing go modules might be the way to go for the future.

dh1tw, 2019-11-14:
Hi Kevin,
sorry that it took some time, but this issue is now solved. I have fixed the dependencies with go modules. You should now be able to build remoteAudio with go 1.13. I have also published [a new release v0.4.0](https://github.com/dh1tw/remoteAudio/releases) with binaries for all common platforms.

closed 2019-11-14 by dh1tw

K6TD, 2019-11-15:
Thanks Tobias.  remoteAudio now builds.  Haven't tested completely, but, remoteAudio enumerate did return a (correct) list.  Tks.

dh1tw, 2019-11-15:
Great! Don't hesitate to get in touch with me when you questions or problems. I'm always happy to receive feedback.