dh1tw/remoteAudio #24: Cross platform audio

bennybubble, 2020-06-10:
Hi Tobias,

Is it possible to get audio across two platforms?
The server side is running on a RPI4 (audio from loopback interface HW:2,1) and the client is on a WINDOWS10 machine (WSAPI : default speakers).
Meanwhile I've tried almost everything but cannot get it to work, I see the client is connecting to the server but no audio is noticable.

Do you (or somebody else) has any experience with this?

-ben

dh1tw, 2020-06-13:
Hi Ben,
I understand that this is a duplicate of #23. 
Yes, remoteAudio absolutely works across multiple platforms. From issue #23 I understand that you were able to resolve the problem setting the sampling rate from 44.1kHz to 48kHz.

If you still experience problems, please feel free to re-open this issue.

bennybubble, 2020-06-14:
Hi Tobias,

I have many problems getting it to work with RPI4 --> WIN10 setup.
I've tested it with RPI4 --> Ubuntu 20.04 without any problems, every time I start the client, the audio is there and solid as a rock!
Using WIN10 I managed it only once to get the audio over the speakers and after that tried to make the output device a VAC. This did not work so I went back to the speakers but it did not work anymore, whatever I try now, no audio anymore...
Are there other ways of testing/debugging ?

-ben

dh1tw, 2020-06-14:
Hey Ben,
I haven't tried to setup remoteAudio on Windows as a VAC. Windows and it's Sound APIs has been a very painful journey for software developers. WASAPI is the most modern API, but doesn't always work properly.
Best supported is MME (however latency can sometimes be a bit high). But it should get you started.
-> Start with MME
-> Make sure sampling rates in remoteAudio reflect the ones of your device
-> Use large buffers in the beginning and long latency values
-> Once this works, decrease buffers and sampling sizes or try another audio API.

how?
Tobias

bennybubble, 2020-06-15:
Hi Tobias,

Thanks for your help.
Using MME did the trick 👍 , I can hear the audio from the app on the RPI4 on my WIN10 speakers, awesome!
There is only one hurdle to take and that has to do with the number of channels:
Working setup with audio redirected to default device (SPEAKERS) : 
`.\remoteAudio.exe client nats -u 192.168.30.23 --output-device-hostapi=MME --output-device-latency=40ms -Y=SDR -R=20 -f=960`
When I want to redirect to audio to [**VAC**](https://vac.muzychenko.net/en/) , it complains about the number of audio channels:
`.\remoteAudio.exe client nats -u 192.168.30.23 --output-device-hostapi=MME --output-device-latency=40ms -Y=SDR -R=20 -f=960 --output-device-name="Line 1 (Virtual Audio Cable)"`
Response:
`unable to open playback audio stream on device Line 1 (Virtual Audio Cable): Invalid number of channels`
I can force both remoteAudio and VAC to 1 or 2 channels but none of the combinations work and always results in the same error message.

Any thoughts ?
Thanks in advance!
-ben

dh1tw, 2020-06-18:
Hi Ben,
I would have to dig a bit deeper into this issue. In principle, you can specify the amount of input and output channels in remoteAudio. In case the the channel number doesn't match between the input and output channels, remoteAudio takes care of removing / duplicating the channel.

If I remember correctly, portAudio (the underlying cross platform library which remoteAudio uses in order to read/write from the sound devices) allows interleaved and non-interleaved audio streams [see the illustration "multi channel wav file"](http://www.labbookpages.co.uk/audio/wavFiles.html).

Maybe searching for portAudio + VAC might also be a pointer into the right direction.

bennybubble, 2020-06-19:
Hi Tobias,

Thanks again for the information but unfortunately I'm not able to get it up and running. I figured out that when using WASAPI as hostapi, remoteAudio doesn't complain about the wrong number of channels but then it just doesn't work using VAC.
Meanwhile I tried other Virtual Loopback drivers and remoteAudio reported the same problem (invalid number of channels). It is a pitty because this would have solved all my problems :-) ! 

It seems to be a problem with virtual audio loopbacks, directing the audio to speakers works fine!

-ben
-ben

bennybubble, 2020-06-19:
Hi Tobias,

Did some further testing and found out that when using WASAPI as hostapi, there is no 'invalid number of channels' error. Redirecting the output to VAC gives gives random results, sometimes there is audio and sometimes there isn't!. About 1 out of 25 attempts it works (when starting remoteaudio client). I've tried many combinations with latency and and rx-buffer-length but for getting the initial connection this doesn't matter. As soon as the connection stops, there is no repeatable way of getting it back again except keep-trying ... :-).

Unfortunately this is not a reliable way of setting up a connection, if you can think of anything please let me know and I'll check it !

-ben

dh1tw, 2021-11-09:
Hey Ben,
I recently released v0.5.2 which also comes with the latest version (19.7) of portaudio. From what I have seen, this release fixes a lot of Windows releated bugs. You might want to give it a try again.

bennybubble, 2021-11-09:
Tnx, I will have a look next weekend!

closed 2021-12-09 by dh1tw