dh1tw/remoteAudio #25: doc for protocol between server and clients?

K6TD, 2020-08-06:
I have a bench top configuration sometimes working.  Occasionally, audio stopping coming out of the client.  If I kill the client process, and restart it, and I click RX on the web page, audio comes back.

As usually, trying to diagnose this problem, I'm trying to learn is it the client, or the server.  It would help if there is some  insight into the protocol flowing between the two thru NATS.  I have nats started with -V.

Or, would I _RTFSC_?
de K6TD

dh1tw, 2020-08-08:
Hey Kevin,

It's pretty easy to subscribe to the respective topics on your NATS broker. The [nats go client](https://github.com/nats-io/nats.go) comes in the example directory with a cli program to subscribe to arbitrary topics. The docs come also with a nice [tutorial](https://docs.nats.io/developing-with-nats/tutorials/pubsub). But it's probably less than 20 lines of code to write your own client.

To get a copy of all the messages passing through your nats broker, you can start it with the `-DV` flag.

For general (performance) information of your nats broker you might want to check out [nats-top](https://docs.nats.io/nats-tools/nats_top).

remoteAudio server listen on the following topic for audio from client(s) - (the audio you want to transmit through your radio):
`shackbus.radio.<server name>.audio.tx`

and the audio from the server (what you receive on the physical radio) is published on the topic:
`shackbus.radio.<server name>.audio.rx`

`<server name>` is specified through the CLI Flag `--server name` or the respective entry in the config file.

It would be interesting to reproduce your problem. Does the log spit out any useful information?

You can btw listen with several remoteAudio clients on the same server (thanks to the NATS pub/sub architecture). I once had a problem with a Logitech USB headset which would randomly stop playing the audio, typically after 30 - 60min. 

Is the connection the broker stable? I think if that connection is lost, the remoteAudio client quits and need to be restarted. Typically a scheduler like systemd will take care of this. 

Starting / Stopping the audio streams through the web interface simply executes an RPC call on the server. Since several clients could be listening to the same server, the server will continue streaming if a client "dies" or goes away without explicitly turning off the audio stream.

Everything (and more) you can do through the web interface, you can also do through the [REST interface](https://github.com/dh1tw/remoteAudio/wiki/Integrate-with-3rd-party-Software-(REST-Interface)). That's a perfect way to integrate remoteAudio with third party applications.

K6TD, 2020-08-09:
Tobias - thanks for the info and hints.  I've set this up as bench top.  Everything is on the same ethernet hub.  No indication of nats connectivity problems. 

I'm logging the messages passing thru nats server . I was looking for the protocol between the server and client remoteAudio. That would allow me to pin down where the audio stops - server or client.  

I'll look see if multiple clients die at the same time.

Thanks.

KR

dh1tw, 2020-08-09:
Hi Kevin,

remoteAudio is using [protocol buffers](https://developers.google.com/protocol-buffers) for serialisation and deserialisation. 

You can find [here](https://github.com/dh1tw/icd/blob/9f0c9c137993ee96a10c0a904953713d45d12065/audio.proto) the server RPC service definition and the corresponding message types.

So, unfortunately sniffing and decoding is not so trivial. I think if you really want to get into the content of the messages, then compiling remoteAudio and adding a few debug statements might be much faster.

Yeah, if you have everything in the local lan, network dropouts shouldn't occur at all. Out of curiosity, which platforms / operating systems are you using? 

I presume you are also using the [latest - 0.4.1 release](https://github.com/dh1tw/remoteAudio/releases)

73 Tobias

K6TD, 2020-08-09:
Tobias,
I'd seen the usage of protocol buffers, Thanks for the pointers to where the RPCs are defined.  The helps.

I realize looking at the messages on NAT isn't trivial.  The server and client are running on an RPi 4, using buster, from May 27. NATS is on a x86 Ubuntu 18.04 LTS.

Yes, using the latest in githhub.

73, K6TD

K6TD, 2021-05-22:
You can close this bug.  I've learned enough from your comments.  And, I've used remoteAudio, across a WAN.

dh1tw, 2021-05-22:
Hey Kevin, 

thanks for the feedback. Have you been able to pin down the root cause of why the audio stopped occasionally? 

Just out of curiosity, do you keep using remoteAudio or did you settle on a different solution? 

73 Tobias

K6TD, 2021-05-23:
I use remoteAudio only.   Used it thru two contests with a pair of flex 6700s.  audio flows consistently now.  I suspect it was configuration issues.

K6TD, 2022-04-11:
Nailed the audio drop out problems. It was an interesting problem.  remoteAudio server and NATs are behind a firewall.  The client runs a VPN to connect to the audio server and the NATs server.  We found the VPN was sometimes not getting packets thru.  Turns out, the VPN, for the tun/tap device on Linux, picks a random MAC address.  In getting several clients created, I duped the config file, for the VPN client.  The "random" MAC address is actually stored in the config file.  Duplicating that file thru several clients created duplicate MAC addresses at the NATS server, resulting in packets not going to the correct tun/tap occasionally.
Once resolved, clients had armchair copy for as long as they wanted to listen.

closed 2022-04-11 by K6TD

dh1tw, 2022-04-11:
Wow - I'm sure that you must have spent quite some time debugging this problem. Congrats - and thanks for the writeup! I hope someone in the future with a similar problem finds this information useful!
I'm sure you have your reasons for the VPN, but I just want to mention that [nats](https://nats.io/) itself supports TLS encryption. With a nats-server at your home(s) and a nats-server at your station you could create a [cluster](https://docs.nats.io/running-a-nats-service/configuration/clustering) or [leaf node](https://docs.nats.io/running-a-nats-service/configuration/leafnodes) setup. The nats-servers takes care of replicating the traffic amongst the nodes.

K6TD, 2022-04-11:
VPN…we took the approach of putting everything behind the FW, and VPN in.  remoteAudio is only one item of access. We have radios, amps, etc.

We deploy the entire setup to several ops.  The less I ask them to setup, the easier they can get on the air (and, I get less IT calls).

thanks Tobias!

KR