dh1tw/remoteAudio #26: Compilation issues

crazybyte, 2021-01-06:
Hello,

when compiling the following error is encountered
make all
...
go: finding github.com/micro/mdns v0.3.0
go: github.com/micro/mdns@v0.3.0: unknown revision v0.3.0
go: error loading module requirements
protoc --proto_path=./icd --gofast_out=./sb_audio ./icd/audio.proto
protoc --proto_path=./icd --micro_out=./sb_audio ./icd/audio.proto
protoc-gen-micro: program not found or is not executable
--micro_out: protoc-gen-micro: Plugin failed with status code 1.
make: *** [Makefile:14: build] Error 1

I figured perhaps it needs some dependencies so I ran make install-deps hoping that it will solve the issue. Unfortunately I get  a very similar error related to github.com/micro/mdns

go: finding github.com/micro/mdns v0.3.0
go: github.com/micro/mdns@v0.3.0: unknown revision v0.3.0
go: error loading module requirements
protoc --proto_path=./icd --gofast_out=./sb_audio ./icd/audio.proto
protoc --proto_path=./icd --micro_out=./sb_audio ./icd/audio.proto
protoc-gen-micro: program not found or is not executable
--micro_out: protoc-gen-micro: Plugin failed with status code 1.
make: *** [Makefile:14: build] Error 1

It seems that is related to some dependencies in micro. The github repo referenced is archived since April 2020. Any fix to circumvent and use something else as dependency to be able to compile the application?

Thank you

dh1tw, 2021-01-06:
Hi @crazybyte,
go-micro is the underlying communication / micro-services framework which is used by remoteAudio. The go-micro project suffered during the last months from several moves, rebranding, reverted rebounding and merges caused by the alternating mood of the author. By today, go-micro: 
- has been moved from [github.com/micro](github.com/micro) to [https://github.com/asim/go-micro](https://github.com/asim/go-micro)
- the former [plugins, examples and protoc-gen-micro repo has been merged](https://github.com/asim/go-micro/discussions/2099) into [https://github.com/asim/go-micro/plugins](https://github.com/asim/go-micro)

I think it's just a matter of sorting out the new locations and updating the dependencies. 

I'm right now pretty busy, but hopefully on the weekend or next week I should find some time to update the dependencies. Otherwise a PR would also be greatly appreciated.

crazybyte, 2021-01-06:
Hello,

I see. Thank you for your reply. I though that probably you are pretty are busy, but I thought that I ask before making a mess of it :) It's not a pressing issue. I will also try my hand in making a fix and a PR, hopefully with some success.  I will leave the issue open (I hope you agree but you can also close it if you want to) so that I can reference it if I manage to make a fix and PR. Thank you again.

dh1tw, 2021-04-17:
Hey @crazybyte, 

I'm happy to report that the dependency issues have been resolved. The Code compiles now again on all supported platforms. Feel free to re-open this issue if you still encounter a problem.

closed 2021-04-17 by dh1tw