dh1tw/remoteAudio #27: no audio when multiple servers

K6TD, 2021-05-22:
When I start two servers (remoteAudio server nats), no audio flows to a client.  If I stop either of the servers, then clients get audio from the remaining server.  
Both servers show to a client on the web page (remoteAudio client nats), and each server can be selected for RX audio.

This is using the prior release code.  Haven't tested since rice was replaced with embedded.  (I have to go visit all the client units to replace the code).

dh1tw, 2021-05-22:
Hey Kevin, 
thanks for the feedback. I haven't used more than one server for quite some time. It's likely that a bug might have crept into the source code. I will try to reproduce the bug and come back to you.

Tobias

K6TD, 2021-05-22:
more testing...
Perhaps a timing issue.  Doesn't happen always.  I found I can also restart the client and now audio flows, once an RX source (server) is selected.

dh1tw, 2021-12-09:
Hi Kevin,
I'm trying to reproduce your problem. I therefore added to my existing remote setup an additional raspberry pi with a USB Headset and my MacbookAIr with the building soundcard as additional servers. Both running remoteAudio v.0.5.2. Unfortunately, I'm not able to reproduce the described problem. Once the servers are discovered by the client, I can flawlessly change the between them without ever losing the audio.  
Are you still experiencing the problem? Does it happen randomly or after a certain amount of time has passed by?

K6TD, 2021-12-09:
what OS/release on the RPi?
I will test again, to see if it's still happening.

dh1tw, 2021-12-09:
- the TS480 uses an OrangePiZero (32bit) with Armbian (based on Ubuntu 18.04)
- the one I added just now is actually an OrangePiPC2 (arm64) with Armbian (based on Ubuntu 20.04)
- MBA is an M1 (arm64) with the MacOS 12.0.1
I will try to add a windows based remoteAudio server. Windows is the least well tested platform. But I assume you are anyway not using windows, right?

K6TD, 2021-12-09:
No windows.  RPI4, or MacOs.
with Ubuntu 20.04 on RPI4 (ARM64), I' having trouble getting alsa to recognize the USB sound card.  probably modprobe problem, as lsusb shows the device and /proc/asound shows a card present.

dh1tw, 2021-12-10:
hmm, strange. I never had an issue with linux / Alsa not recognizing a USB soundcard - and over time I have tried many. From 1USD china crackers to premium Logitech and Beyerdynamic ones.
I'm btw in the middle of releasing v0.5.3 which fixes a bug I found recently - remoteAudio not accepting other samplingrates  than 48kHz. Now also 44.1kHz input devices will work properly.

K6TD, 2022-04-11:
K6TU has helped resolve a number of problems.  I don't see this issue any more.

closed 2022-04-11 by K6TD