dh1tw/remoteAudio #30: just struggling

Littlechay, 2021-08-13:
HI. 

I have remote raspberry pi (a 3b in this case) with a Yeasu FT857d attached via a generic USB sound card. in my quest to find a good method of providing 2-way audio from my W10 pc to the rig I came across remoteAudio. Everything looks very sane to me, it is the way I want to go, but I just can't get it work. Not a squeak in either direction, no web controls - all I get is a 404 error from the server?

I have NATs installed on the Pi. It runs without error
I have remote audio on the pi and run it as `./remoteAudio server nats` - also runs without error
I have the config file on the pi as per the one in the repository with the only changes being the rig name and bumping up the buffering to 100ms

On the windows machine I am running remote audio `.\remoteAudio.exe client nats --config .remoteAudio.toml`
It appears to run fine - output as follows. 

```
> .\remoteAudio.exe client nats --config .remoteAudio.toml
Using config file: .remoteAudio.toml
2021/08/13 23:30:39 output sound device: CABLE-B Input (VB-Audio Cable B), HostAPI: Windows WASAPI
2021/08/13 23:30:39 input sound device: CABLE-A Output (VB-Audio Cable A), HostAPI: Windows WASAPI
2021/08/13 23:30:42 added audio server ft857d
2021/08/13 23:30:42 webserver listening on localhost:9090
```
The config file is as per the on in the repository but with beffers to 100ms and the nats ip set to that of the pi

The sound card works. I can run rtp over it . 
Everything is set to 16bit 48000Hz as far as I can tell. 

I have VACs on the windows machine running into JTDX but have also tested to speakers etc.. 

Audio setup on PCs and streaming in gneral,  is not something I know much about. I'd be grateful for any pointers on how to start debugging my setup. 

Cheers
Chris ZL1OSC, VP8DSC, ex VP8BKF

Littlechay, 2021-08-14:
Further update. I have got things running for no apparent reason except that I started with  Pi 4 and a clean install. I may actually have had something working before but not realised as the audio level, at least in the pi to PC direction are so low that it is only audible with AKR (all knobs right). I know the levels are OK as it works via rtp.. Any ideas anybody? 

So I know the audio level into the pi's usb card is OK, and OK in Alsa,  and I know that it's not OK out of remoteAudio on the Windows PC. 

Changing the default volume settings in .remoteAudio.toml or with p commands does very little suggesting that there is very little to work with , or compression occuring, over NATS

Littlechay, 2021-08-15:
OK getting there. now sorry for the noise. 

I can't get remoteAudio to connect my virtual audio cable (well nothing at all now in fact!) I wasn't having success VB audio cable so downloaded VAC but the results are exactly the same. Invalid number of channels.... Oh that and still no web interface! 

I see this number of channels thing has been reported before in issues but I couldn't find a solution. 

```
.\remoteAudio.exe client nats --config .remoteAudio.toml --output-device-hostapi=WASAPI --stream-on-startup --server-name ft857d
Using config file: .remoteAudio.toml
2021/08/15 16:46:36 output sound device: Line 1 (Virtual Audio Cable), HostAPI: Windows WASAPI
unable to open recording audio stream on device Line 2 (Virtual Audio Cable): Invalid number of channels
PS C:\Users\64210\Documents\Ham Radio>
```

Littlechay, 2021-08-16:
OK Another update. I can always eventually get remoteAudio and VAC to connect without the "Invalid number of channels" error by toggling the speaker pin on the VAC facing remote audio off then on. To get it to connect the first time I had to toggle spkr on-off-on and subsequently off-on at least once. After a couple of reboots it seems stable. Only issue is that W10 sees the cable as a speaker so configuration is a bit odd! So all working ok except no web interface. Would love to get that up and have easy access to the volume controls. Any pointers from anybody please? Could be a path issue? What path should the index.html be on?

peterus, 2021-08-18:
which address are you using for the webpage? for you it should be the ip from the windows machine.

Littlechay, 2021-08-18:
I'm using the correct address. I've been doing some investigating. I believe the problem lies in the way the paths are encoded /zzzzz/xxxx/ instead of the windows \zzzzzz\xxxxx

dh1tw, 2021-08-20:
Hi Chris, 
the missing web interface is unfortunately a bug on the windows build. Since remoteAudio is a single static binary file, I include the HTML, CSS, JS... files during compilation. Since Go 1.16 this is natively supported. There must be something wrong in the windows build pipeline. A few days ago Marios, 5B4WN reported the bug as well. 

Until I have provided a bugfix, you can interact with the remoteAudio client through it's [REST API](https://github.com/dh1tw/remoteAudio/wiki/Integrate-with-3rd-party-Software-(REST-Interface)) if you don't mind using [curl](https://curl.se/) or [httpie](https://httpie.io/) from the command line.

dh1tw, 2021-11-07:
Solved with [remoteAudio v0.5.2](https://github.com/dh1tw/remoteAudio/releases/tag/v0.5.2)

closed 2021-11-07 by dh1tw