dh1tw/remoteAudio #34: Raspberry Pi (bullseye): Segmentation fault

MichaelRenner, 2022-01-09:
Moin,

I set up a a new installation at an old RasPi Model B Plus Rev 1.2 (but I get the same result with a brand new Raspberry Pi Zero) and unpacked remoteAudio-v0.5.3-linux-arm.zip

```
root@alschain:~# remoteAudio 
Segmentation fault
root@alschain:~# uname -a
Linux alschain 5.10.63+ #1488 Thu Nov 18 16:14:04 GMT 2021 armv6l GNU/Linux
root@alschain:~# file /usr/local/bin/remoteAudio 
/usr/local/bin/remoteAudio: ELF 32-bit LSB executable, ARM, EABI5 version 1 (GNU/Linux), statically linked, no section header
```

Any hint?

Thanks from Micha, DD0UL

dh1tw, 2022-01-10:
Hi Michael,

unfortunately, I don't have an old Raspi1 or Zero model here to reproduce your problem. A couple of users have reported performance problems with single core Raspis, that's why I excluded the old single core Raspis from the [list of supported devices](https://github.com/dh1tw/remoteAudio#supported-platforms). 

Anyway, a SegFault right at the beginning is strange. I'm not sure how I can help you.

Have you tried remoteAudio on a bit more modern platform like a Raspi 2 or Raspi 3? I'm personally using remoteAudio with great success with OrangePiZeros on a daily basis.

MichaelRenner, 2022-01-11:
Moin Tobias,

thank you for your answer. New Rasp Pi are currently hard to find (200 days delivery time or almost 200 Euro), so I took an old Rasp Pi out of my craft box (and bought the brand new RasPi Zero).

RemoteAudio starts on an other RasPi 4, but this is integrated in another project. If you want to take a deeper dive I can set up a login for the RasPi Zero for you. If you don't like it too, I'll think about a plan B :-)

vy 73 de Micha, DD0UL

K6TD, 2022-01-11:
what is the OS being used?  And, Tobias - when you test, are you using Raspbian, or Ubuntu, on a RPI4?

dh1tw, 2022-01-12:
I'm running on all my embedded Devices [Armbian](https://www.armbian.com) which is based on Debian / Ubuntu. The Armbian versions I'm using are at least based on Ubuntu 18.04 (Kernel 4.19).

MichaelRenner, 2022-01-12:
Moin, I'm using Raspbian, Armbian does not support Raspberry Pis. However: Due the fact that I had no success debugging the application (and, as you mentioned, users reported performance problems with single core RasPis, my next step is to search for alternative hardware.

```
pi@auriga:~ $ gdb /usr/local/bin/remoteAudio 
GNU gdb (Raspbian 10.1-1.7) 10.1.90.20210103-git
Reading symbols from /usr/local/bin/remoteAudio...
(No debugging symbols found in /usr/local/bin/remoteAudio)
(gdb) run
Starting program: /usr/local/bin/remoteAudio 

Program received signal SIGSEGV, Segmentation fault.
0x0031cfdc in ?? ()
(gdb) backtrace
#0  0x0031cfdc in ?? ()
#1  0x0033969c in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) 
```
vy 73 de Micha, DD0UL

K6TD, 2022-01-12:
I’ve had luck running Ubuntu on RPI 4B, and remoteAudio.  It takes some setup.  Armbian is getting a build working for RPI 4B.  Raspbian seems to have too many issues, for this kind of application.

73, K6TD

dh1tw, 2022-01-12:
I have the suspicion that this problem is related to the 256Gbyte RAM available on the Raspi1 and RaspiZero. When creating a release, I compress the binary with [upx](https://upx.github.io/). That's the reason why gdb can't find an ELF header. I can only imagine that the decompression happens in memory on the fly and then maybe causes a SegFault. 
I just created manually a [release for arm6 without compression](https://cloud.dh1tw.de/s/pnPgDpRF9zTBr2K). Please give it a try.

MichaelRenner, 2022-01-13:
Thank you for this special version. remoteAudio still crashes at this 512MB single core RasPi, but we have an hint now:

```
pi@auriga:~ $ gdb ./remoteAudio 
GNU gdb (Raspbian 10.1-1.7) 10.1.90.20210103-git
Reading symbols from ./remoteAudio...
(gdb) run
Starting program: /home/pi/remoteAudio 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x003314c8 in type..eq.internal/cpu.option (p=0xbefff71d, q=0xbefff731, r=<optimized out>) at <autogenerated>:1
1       <autogenerated>: No such file or directory.

pi@auriga:~ $ cat /proc/cpuinfo 
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 997.08
Features        : half thumb fastmult vfp edsp java tls 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2835
Revision        : 9000c1
Serial          : 00000000a798dedd
Model           : Raspberry Pi Zero W Rev 1.1
```

vy 73 de Micha, DD0UL

dh1tw, 2022-01-16:
That line with `eq.internal/cpu.option` looks a bit suspicious. If I remember correctly, support for some very old ARM5 and ARM6k platforms has been dropped, but I always thought that the Raspberry Pi (1) wasn't affected. 
Can you check if this [helloworld](https://cloud.dh1tw.de/s/6MepqJ6WXtJK3qi) generated with go 1.17 program executes?

MichaelRenner, 2022-01-17:
Moin,

the helloworld is running at the Raspberry Pi Zero as well as at the Raspberry Pi Model B Plus. 

```
pi@auriga:~ $ unzip helloworld-linux-arm.zip 
Archive:  helloworld-linux-arm.zip
  inflating: helloworld              
pi@auriga:~ $ ./helloworld 
Hello World

pi@alschain:~ $ unzip helloworld-linux-arm.zip 
Archive:  helloworld-linux-arm.zip
  inflating: helloworld              
pi@alschain:~ $ ./helloworld 
Hello World
pi@alschain:~ $ cat /proc/cpuinfo 
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 697.95
Features        : half thumb fastmult vfp edsp java tls 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2835
Revision        : 0010
Serial          : 000000000eebcf3b
Model           : Raspberry Pi Model B Plus Rev 1.2
```

Greetings