dh1tw/remoteAudio #36: Output volume at radio from server is very low

pkubat, 2022-03-15:
Sorry to open another issue, but I did not want to hijack the other thread.  Also, I am not 100% convinced this is an remote audio issue, but need a little help to diagnose it.

```
+--------+                           +-----------------+
|        |       audio cable         | raspberrypi     |
| radio  |  <- usb 'sound card' ->   |  w/ remoteAudio |
|        |                           |  server         |
+--------+                           +-----------------+

                                            ^
                                            |
                                            v

                                        +--------+
                                        |  nats  |
                                        | docker |
                                        +--------+

                                            ^
                                            |
                                            v

+-------+       +--------------+     +-------------+
|       |       |   virtual    |     |  macOS      |
| jtdx  |  <->  |   loopback   | <-> | remoteAudio |
|       |       | audio device |     |  client     |
+-------+       +--------------+     +-------------+

```

* lastest build of remoteAudtio
* tried a new build of master, but had other issues (will retry again later)
* tried [VB-Cable](https://vb-audio.com/Cable/index.htm) and [Loopback](https://rogueamoeba.com/loopback/) for the virtual cable; both have the same issue.
* Loopback allows for "monitoring" of the audio channel and they seem fine.  So the audio from Jtdx seems fine.
* All the hardware is fine, as it works if I connect everything to the mac
* I remoted the radio and used two sets headphone and you can bearly hear the output
* The volumes level in `.remoteAudio.toml` are 100%
* jtsx is at 0db if directly connected usually about ~4db
* Also the quality of the audio has static (I see there was a fix in new code, this the new build)
* need to test more, but on the first pass, it seemed the audio from "Builtin Mac Mic" was okay, though hard to tell being on person at the same time.

I am open to suggestions of what to try next.  Do your use digital modes?

Thanks!

dh1tw, 2022-03-16:
Hey @pkubat, 
makes sense to start a new thread. Have you already adjusted the audio levels on the raspberry Pi? Sounds like your line-in or microphone levels on the Pi are too low. You can use the Alsamixer to adjust the volume of each audio device. Try `sudo alsamixer` and then persist the settings with `sudo alsactl store`. 

You could also record a sample on the raspi with `arecord` and then either replay them locally on the Raspi with headphones or via scp on your Mac. MacOS comes with a nifty commandline tool called `afplay`. 

I'm not really a digital mode operator, but a few months ago I did the same tests (WSJT-X / FT8) with my Mac and a Windows PC. Both worked well and audio levels were fine. 

Could the "static crashes" be audio dropouts? Try increasing the audio packet size.

pkubat, 2022-03-19:
@dh1tw Sorry my mistake!  I did not save the volume settings, even worse in that you took the time to document the entire process in the [wiki](https://github.com/dh1tw/remoteAudio/wiki/Audio-Devices#helpful-alsa-cli-tools) I need to learn to read!

So the good news is that it is "working", though the quality of the audio needs need work.  I'll find time in the next day or so to tweak the settings and see if I can resolve.

I did have a few questions on the configuration.
1. What are the best defaults?  [`.remoteAudio.toml`](https://github.com/dh1tw/remoteAudio/blob/master/.remoteAudio.toml) and `remoteAudto -h` have different defaults.  So to have a good starting point.
2. What settings apply to the "client" and which ones are for the "server"?
3. Is there any document on what each setting does?  (if not I can try to take a pass at creating some)

Again thanks!!!!!

dh1tw, 2022-03-20:
Glad to hear that the wiki was of help. 
The easiest way to check the available parameters for a remoteAudio client and server is to execute `./remoteAudio server nats --help` and `./remoteAudio client nats --help` respectively. The priority is as follows:
1. command line flags (highest)
2. config file 
3. default values (lowest)

I just updated the default config file [.remoteAudio.toml](/dh1tw/remoteAudio/blob/master/.remoteAudio.toml) and added explanations to the parameters. I hope this gives you a better understanding of the individual parameters and where they are used. 

In principle, the default values should work well and get you started out of the box. Once you feel comfortable, you can start tweaking the parameters.

I hope this helps!

73 Tobias