dh1tw/remoteAudio #37: panic: runtime error: invalid memory address or nil pointer dereference

MichaelRenner, 2022-04-01:
Moin,

Server 1: Raspberry Pi 3 Model B Plus Rev 1.3 with Raspbian 11.3 and USB Logitech Headset (also tested with simple USB Sound Adapter)
Server 2: AMD64 Notebook with Debian 11 and internal Sound
Client: Ubuntu with pulseaudio
Nats-Server: 2.7.4 in the same network

Receiving is working fine.
As soon as a  press the PTT button serverside remoteaudio crash:

```
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x18 pc=0x10028be]

goroutine 2131 [running]:
github.com/dh1tw/remoteAudio/audio/sources/pbReader.(*PbReader).Enqueue(0xc0002ff020, {0xc00003a948, 0x12, 0x12})
        /usr/src/myapp/audio/sources/pbReader/pbReader.go:144 +0x3be
github.com/dh1tw/remoteAudio/cmd.(*natsServer).enqueueFromWire(0xc0000ae0e0, {0x15b2d70, 0xc0001283f0})
        /usr/src/myapp/cmd/server_nats.go:395 +0x53
github.com/asim/go-micro/plugins/broker/nats/v3.(*natsBroker).Subscribe.func1(0xc00010eae0)
        /go/pkg/mod/github.com/asim/go-micro/plugins/broker/nats/v3@v3.7.0/nats.go:224 +0x2a9
github.com/nats-io/nats%2ego.(*Conn).waitForMsgs(0xc0003ae600, 0xc00032c180)
        /go/pkg/mod/github.com/nats-io/nats.go@v1.13.0/nats.go:2667 +0x442
created by github.com/nats-io/nats%2ego.(*Conn).subscribeLocked
        /go/pkg/mod/github.com/nats-io/nats.go@v1.13.0/nats.go:3882 +0x3f1
```

This happens in every combination (server / soundcard)

Server:
pi@alschain:~ $ remoteAudio server nats -Y "PuxingPX2R"  -i 'Logitech USB Headset: Audio (hw:1,0)' -o 'Logitech USB Headset: Audio (hw:1,0)' -u hermes.fritz.box

Client:
renner@solaris:~$ remoteAudio client nats -Y "PuxingPX2R"  -i 'pulse' -o 'pulse' -u hermes.fritz.box

I am lost. Any hint?

dh1tw, 2022-04-01:
Hi Michael, 
just to be sure, have you updated both remoteAudio instances to the latest version (0.5.4)? 

Tobias

MichaelRenner, 2022-04-02:
Hi Tobias,

actually on both devices the version was 0.5.4. Tested twice. Now I have installed version 0.5.3 on both computers - and the error does not occur. That's why I'm using this version now and I'm looking forward to version 0.5.5 ;-)
If I can help you with debugging: write me a DM.

vy 73 de Micha, DD0UL

dh1tw, 2022-04-04:
I was able to reproduce and fix the problem. Should be solved with [v0.5.5](https://github.com/dh1tw/remoteAudio/releases/tag/v0.5.5)

K6TD, 2022-04-11:
FWIW, I ran into this on 0.5.4, and found two fixes ...use 0.5.5, or fill in the user= parameter in the [NATS] section info the .remoteAudio.toml file.  
Nice added benefit.  When a user enables TX, the userid is sent to all clients.