dh1tw/remoteAudio #38: Error when using Apple Airpods as output device on Mac M1

hamav8tor, 2022-12-31:
Any ideas? Works fine on Mac speaker output.

$ remoteAudio enumerate

Available audio devices and supported Host APIs:

	Detected 1 host API(s):

	Name:                   Core Audio
	Default input device:   'MacBook Air Microphone'
	Default output device:  'MacBook Air Speakers'
	Devices:
		Name:                      'MacBook Air Microphone'
		MaxInputChannels:          1
		MaxOutputChannels:         0
		DefaultLowInputLatency:    33.6875ms
		DefaultLowOutputLatency:   10ms
		DefaultHighInputLatency:   43.020833ms
		DefaultHighOutputLatency:  100ms
		DefaultSampleRate:         48000

		Name:                      'MacBook Air Speakers'
		MaxInputChannels:          0
		MaxOutputChannels:         2
		DefaultLowInputLatency:    10ms
		DefaultLowOutputLatency:   24.416666ms
		DefaultHighInputLatency:   100ms
		DefaultHighOutputLatency:  33.75ms
		DefaultSampleRate:         48000

		Name:                      'Microsoft Teams Audio'
		MaxInputChannels:          2
		MaxOutputChannels:         2
		DefaultLowInputLatency:    10ms
		DefaultLowOutputLatency:   1.333333ms
		DefaultHighInputLatency:   100ms
		DefaultHighOutputLatency:  10.666666ms
		DefaultSampleRate:         48000

$ remoteAudio client nats --config remoteAudio-bt.toml
Using config file: remoteAudio-bt.toml
unable to open playback audio stream on device Dave’s AirPods Pro: Invalid number of channels

$ cat remoteAudio-bt.toml
[nats]
broker-url = "home-shack"
broker-port = 4222
username = "km3t"
password = "xxxxxxxxxx"

[input-device]
device-name = "default"
samplerate = 48000
latency = "5ms"
channels = 1
hostapi = "default"

[output-device]
device-name = "Dave’s AirPods Pro"
samplerate = 48000
latency = "5ms"
channels = 1
hostapi = "default"

[opus]
application = "restricted_lowdelay"
bitrate = 24000
complexity = 5
max-bandwidth = "wideband"

[audio]
rx-buffer-length = 5 # 10x10ms (@framesize=480 & samplerate=48kHz)
# opus does not accept arbitrary frame length! Check OPUS documentation
frame-length = 480  #10ms @ 48kHz (should work for most cases)
tx-volume = 70
rx-volume = 90
stream-on-startup = false
vox = false
vox-threshold = 0.1
vox-holdtime = "500ms"

[http]
host = "localhost"
port = 9090

hamav8tor, 2022-12-31:
Actually, solved just using default. Problem likely stemmed from both devices having the same name.

dh1tw, 2023-01-05:
Hi Dave,

interesting problem. Glad to hear, that using the default device solved the problem. 

Out out curiosity, do you have two sets of AirPods or does each Airpod (Left & Right) show up as an output device?