dh1tw/remoteAudio #41: why it's timeout

Difkyear, 2024-05-06:
`.\remoteAudio.exe server nats --config remoteAudio.toml
Using config file: remoteAudio.toml
2024/05/06 13:59:35 output sound device: Speakers (Realtek(R) Audio), HostAPI: Windows WASAPI
2024/05/06 13:59:35 input sound device: Microphone Array (Intel® Smart Sound Technology for Digital Microphones), HostAPI: Windows WASAPI
2024/05/06 13:59:37 dial tcp 127.0.0.1:4221: i/o timeout`

Difkyear, 2024-05-06:
i use [.remoteAudio.toml] this file in repository,   for config to run

dh1tw, 2024-05-25:
Looks like remoteAudio is trying to connect to the nats broker on localhost, port 4221. Can you check the config file and check the port of the nats broker?

closed 2025-03-05 by dh1tw