dh1tw/remoteAudio #42: segmentation fault

pkubat, 2025-02-13:
trying to run the latest release v0.5.5 on macOS 15.3.1 (24D70)

```
% ./remoteAudio
zsh: segmentation fault  ./remoteAudio
```

I also tried to build from source using https://github.com/dh1tw/remoteAudio/wiki/Build-instructions-MacOS

```
% make build
protoc --proto_path=./icd --micro_out=. --go_out=. ./icd/audio.proto
go build -v -ldflags="-X github.com/dh1tw/remoteAudio/cmd.commitHash=fafc017 \
		-X github.com/dh1tw/remoteAudio/cmd.version=v0.5.5-1-gfafc017"
# github.com/dh1tw/remoteAudio
link: golang.org/x/net/internal/socket: invalid reference to syscall.recvmsg
make: *** [build] Error 1
```

Also, how active is this project?  Are you still using it for your remote shacks?  It does not look like there are a lot of update lately?  BTW, I will help as, but I have limited go knowledge.

Thanks!

dh1tw, 2025-02-17:
Hi @pkubat,

yes, the project is still in active use. So far it does all I need, that's why I spent my time rather on other projects.

I was able to reproduce the build error you described. I'll have a look at it and fix it.

dh1tw, 2025-02-17:
I upgraded the broken dependencies. You should be able to build remoteAudio again straight from the sources. I'm still not sure why you get a seg fault on the v0.5.5 binaries. Maybe it's time to release freshly compiled binaries with go 1.24.

dh1tw, 2025-03-05:
You should be able to build remoteAudio again by yourself. After 3 years, it was time to update the dependencies and create a new release (v.0.5.6) using the latest go compiler (1.24). 
If you still encounter problems, feel free to reopen the issue.

closed 2025-03-05 by dh1tw