dh1tw/remoteAudio #45: github.com/golang/protobuf is deprecated: Use the "google.golang.org/protobuf" module instead.

roelandjansen, 2025-03-06:
(re)building to make things work on suse / sles / leap / tw.  

dumped remoteAudio and re-cloned including the (re)cursed submodules:

roeland@zeetak:~/src/remoteAudio> make install-deps
go mod download
go install github.com/golang/protobuf/protoc-gen-go@v1.5.2
go: module github.com/golang/protobuf is deprecated: Use the "google.golang.org/protobuf" module instead.
go install github.com/asim/go-micro/cmd/protoc-gen-micro/v3@v3.7.0

make build fails but let me find out why before I create an issue or pr.

73s pa3met

dh1tw, 2025-03-06:
have you checked out the latest sources? - I coincidentally released v0.5.6 last night with updated dependencies, including the switch from github.com/golang/protobuf to google.golang.org/protobuf.

roelandjansen, 2025-03-06:
it was cloned 5 minutes before I created the issue.

dh1tw, 2025-03-06:
I see. It's most likely caused by an indirect dependency. I suspect something from github.com/asim/go-micro. I anyway want to replace the transport layer, but that's a larger effort.