dh1tw/remoteAudio #48: remoteAudio-v0.5.7-darwin-amd64 ` bad CPU type in executable: ./remoteAudio`

pkubat, 2025-04-19:
The macos amd does not run on intel mac: `bad CPU type in executable: ./remoteAudio`.  Maybe the arm was bundled? A new build from source works fine.  the AMD and ARM unzipped file are the same via diff.

dh1tw, 2025-04-19:
Thanks for the feedback @pkubat. 

indeed:

```
$ unzip ./remoteAudio-v0.5.7-darwin-amd64.zip
$ file ./remoteAudio
remoteAudio: Mach-O 64-bit executable arm64
``` 

It looks like Github now requires a paid plan for using Darwin/AMD64 in Github Actions. I upgraded my plan and updated the CI. Version 0.5.8 now provides a proper Darwin/AMD64 build. 

I would appreciate it if you could confirm that the v0.5.8 build now works correctly on Darwin/AMD64.

pkubat, 2025-04-19:
looks good!

```
 % ./remoteAudio version
copyright Tobias Wellnitz, DH1TW (2025)
remoteAudio Version: v0.5.8, darwin/amd64, BuildDate: 2025-04-19T15:31:13-04:00, Commit: af65fdc

% file ./remoteAudio
./remoteAudio: Mach-O 64-bit executable x86_64
```

closed 2025-04-19 by dh1tw